entech281/Robot2023 #124: Create autonomonmous far + dock command

dcowden, 2023-03-09:
We need a command that does the same as CommandFactory.autonomousFarCommand(), but that drives the right distance to stop at the middle of the dock.

If we are able to test this effectively, we will also need to add back the autonomous chooser in shuffleboard. Currently it is disabled since we only have one command we want to run.

This version of the command should NOT use any magic numbers

mandrews281, 2023-03-09:
We we add the robot telemetry, the encoders and NavX data should get logged.  We could correlate the pitch angle changes with the drive distance and (potentially) know exactly when to stop after driving over the charging station in autonomous once.

dcowden, 2023-03-09:
the work here is to do these things:

- [ ] calculate the distance we need to travel to dock
- [ ] write a new autonomous command that's the same as autonomousfar ( in CommandFActory), but travels that distance instead
- [ ] re-enable the code in Robot.java that allows choosing the auto routine on shuffleboard
- [ ] test

closed 2023-03-10 by dcowden