entech281/Robot2023 #135: feedback assisted docking

dcowden, 2023-03-10:
We will try dead-reconing docking at anderson. but unless it performs flawlessly, we should add feedback to assist accuracy.

(1) should be usable in both auto AND at end of match
(2) could use gyro and/or vision feedback ( we know how far we are from an april tag pretty precisely

dcowden, 2023-03-12:
this is lower priority than cone scoring work. 
additionally, we had brake mode off, so our existin gmight work
and also, there's some doubt that we have enough friction to stop anyway.
If we do this work, we should also look at a more evolved auto

dcowden, 2023-03-12:
for example, #149

dcowden, 2023-03-12:
the mechanical solution for this is pneumatically actuated presser feet, that will just stop the robot from moving

mandrews281, 2023-03-22:
works.