entech281/Robot2023 #172: use visoin data for lateral positioning only?

dcowden, 2023-03-12:
could it be more useful for lateral control only vs pose estimation?

*can we see the apriltag close enough?
* is lateral data good for corrrection?

dcowden, 2023-03-12:
depends on what we learn in #173

dcowden, 2023-03-14:
the requested functionality is:

when you can see an april tag:
    if one button on the joystick is pressed, move (strafe) sideways to the next scoring location towards the right. 
    if a different button on the joystick is pressed, move (strafe) sideways to the next scoring location to the left

if there is no scoring location in the direction requested, or if there is no april tag in view, do nothing.

the first step is to add Y offest to the shuffleboard, so we can see how acurate it is.

the move should be done with a PID controller, using the desired location as the setpoint.

this issue is for making the command. 

we will also need a class that calculates the next scoring location to the right

dcowden, 2023-03-14:
also note: before we run this command, we'll probably run a set to yaw angle command to align the robot square to the scoring position

dcowden, 2023-03-15:
@WhyDoesGodDoThis a starting point for this is on main. it is not tuned. We can't tune /use it in real life till mike finishes fixing yaw lock.

the next programming task is to change RobotState ( which implmenets LateralOffsetSupplier) to give the offset to the NEAREST scoring location.  the Horizonatal alignment controller will try to drive that offset to zero.

dcowden, 2023-03-15:
NOTE: the center location is where the april tag itself is .  we need to make sure we include the loading stations too. that way, we can align to the loading station

dcowden, 2023-03-15:
also: important requirement:  when we lose vision, but this command is enabled, we probably want to allow the operator to move laterally. that way, they can manually nudge to correct

dcowden, 2023-03-15:
update/suggestion: while there are probalby plenty of really cool algorithms to come up with which scoring location is closest, the simplest and fastest to code is probably to accept the current absolute field loation, and then iterate over all of the locations, returning the one closest

dcowden, 2023-03-16:
we have the command written. what needs to be done now is tuning.

current behavior: it tries to work, but stutters. 
its either:
(a) gain issue
(b) fighting with hold yaw filter
(c) intermittent camera feed

next steps:
come up with a way to get a temporary stable ( non-camera) input signal and see what happens.

mandrews281, 2023-03-22:
scoring locations work.  Need to test loading station locations