entech281/Robot2023 #214: Label driver  joystick buttons

dcowden, 2023-03-19:
We don't want confusion , and especialy we dont want a situation where if we need a backup driver, the driver has to learn all the buttons without a reference! at a minimum, we need a two column sheet with number and function.

John-Molfneter, 2023-03-19:
Trigger = Togle Turn
Thumb Button = Auto Align
Thumb Pad Bottom Left = Nuge Yaw Left
Thumb Pad Bottom Right = Nuge Yaw Right
Button Pad Top Left = Brake/Coast Toggle
Button Pad Bottom Left = Precision Drive
Button Pad Top Middle = Zero Robot Angle
Button Pad Bottom Middle = Zero Gyro Angle
Button Pad Top Right = Field Absolute
Button Pad Bottom Right = Auto Yaw

dcowden, 2023-03-20:
can you include the button numbers for each of these? they rae labelled on the joystick. eample: zer gyro is button 9 i think

mandrews281, 2023-03-22:
Partial labels (base buttons)

closed 2023-03-22 by mandrews281