entech281/Robot2023 #237: Add logic to center autonomous for charging station balance

mandrews281, 2023-03-20:
Decide based on match time if we can go for the movement points by driving over the charging station and returning.  Needs to be something like:
```

public boolean isTimeForCommunityMove() {
   if ( DriverStation.getMatchTime() < 5.0 ) {
      DriveStation.reportWarning("Moving and Balancing",false):
      return true;
  }
  DriverStation.reportWarning("Time for Balance only",false);
  return false;
}

ConditionalCommand(driveOverAndBalanceCommand, balanceOnlyCommand, this::isTimeForCommunityMove);

```
This probably all belongs in the CommandFactory.

dcowden, 2023-03-20:
oh, this is a very clever idea!

dcowden, 2023-03-20:
we also need to run the full center auto routine to see it work-- ideally 3 times in a row.

We can save time still on these areas:
* increase homing move speed
* decrease grippers waits a bit
* move elbow in parallel with beginning to drive

mandrews281, 2023-03-21:
Done as part of autonomous refactoring.  Note: DriverStation.getMatchTime() returns the time REMAINING in the current game phase so it's a countdown (not a countup) timer.

mandrews281, 2023-03-22:
done.  works

closed 2023-03-22 by mandrews281